jamesbowman/swapforth #3: Cannot connect via USB serial at 115200 or even 57600

combinatorylogic, 2015-08-05:
But 38400 works ok. Any ideas what can cause this? Host: stock ubuntu 14.04.

dcwbrown, 2015-08-05:
I had the same problem - didn't try higher speeds, but I did find 2 other ways to fix it.

First I first added time.sleep(0.01) after j1a/shell.py line 30 (ser.flushInput()).

Later, looking at uart.v I noticed the tx code sends 5 stop bits as a result of j1a/icestorm/uart.v line 77 which reads:
   bitcount <= 1 + 8 + 1 + 4;

I can't think why the + 4 is there, and indeed removing the final '+ 4' from fixed the problem (so I didn't need the sleep in shell.py).

(I'm using Ubuntu 15.04.)

combinatorylogic, 2015-08-05:
None of these solutions worked for me

dcwbrown, 2015-08-05:
Does shell.py show 'Contacting...' and then hang?

Have you tried enabling the commented-out print statements on shell.py lines 31 and 36 - you should see all of the ' 1 tth!\r' and the chr(30) echoed character by character - for me the initial ' 1 ' usually worked but everything from tth was corrupted - not always the same way. 

Once '1 tth' has been sent successfully swapforth goes into tethered mode where uart timing is much less sensitive.

combinatorylogic, 2015-08-05:
Yes, it hangs on 'Contacting', RXD goes up once. Guess I'll stick to 38400bps for now.

combinatorylogic, 2015-08-06:
Can confirm that the recent commits fixed the problem. Thanks!

closed 2015-08-06 by combinatorylogic

jamesbowman, 2015-08-06:
Awesome, thanks!