jamesbowman/swapforth #10: Error compiling with Icestorm

polygonhell, 2015-08-30:
I get the following error running make -C icestorm as of Aug 29th
fatal error: failed to place: placed 1125 LCs of 1294 / 1280

I'm not sure if this is an Icestorm issue or a swap forth issue
full output below
make[1]: `build/ram.v' is up to date.
yosys  -q -p "synth_ice40 -top top -blif j1a.blif" j1a.v uart.v ../verilog/j1.v ../verilog/stack2.v
Warning: Wire top._j1.reboot has an unprocessed 'init' attribute.
Warning: Wire top._uart0._rx.hh has an unprocessed 'init' attribute.
Warning: Wire top.unlocked has an unprocessed 'init' attribute.
arachne-pnr -p j1a.pcf j1a.blif -o j1a.txt
seed: 1
device: 1k
read_chipdb +/share/arachne-pnr/chipdb-1k.bin...
  supported packages: tq144
read_blif j1a.blif...
prune...
read_pcf j1a.pcf...
instantiate_io...
pack...

After packing:
IOs          24 / 96
LCs          1293 / 1280
  DFF        730
  CARRY      103
  CARRY, DFF 32
  DFF PASS   97
  CARRY PASS 15
BRAMs        16 / 16
WARMBOOTs    0 / 1
GBs          0 / 8

promote_globals...
  promoted clk$2, 794 / 794
  promoted _j1.rstack.move, 305 / 305
  promoted _j1.dstack.move, 259 / 259
  promoted $techmap$techmap9280$auto$simplemap.cc:449:simplemap_adff$793.$logic_not$/usr/local/bin/../share/yosys/ice40/cells_map.v:12$9274_Y, 36 / 36
  promoted unlocked, 17 / 17
  promoted _j1.rstack.delta[1], 304 / 304
  promoted _j1.dstack.delta[1], 260 / 260
  promoted $techmap\_j1.dstack.$procmux$209_CMP, 16 / 16
  promoted 8 nets
    3 sr/we
    4 cen/wclke
    1 clk
  8 globals
    3 sr/we
    4 cen/wclke
    1 clk
realize_constants...
  realized 1
place...
fatal error: failed to place: placed 1125 LCs of 1294 / 1280

polygonhell, 2015-08-30:
Looks like this is just some difference in the synthesis or layout, it's now taking a few extra LC's and place fails. I reduced the stack sizes and it works.

jamesbowman, 2015-08-30:
Is this with latest yosys and arachne-pnr? For the record I am getting a clean build with these versions:

    $ cd yosys
    $ git rev-parse --short HEAD
    eb38722
    $ cd ../arachne-pnr/
    $ git rev-parse --short HEAD
    a941129

polygonhell, 2015-08-30:
I did a git clone yesterday - looks like I have exactly the same versions as you (see below).
Having said that I'm running the icestorm tools on a Mac, perhaps that's the difference, is the number of LC's dictated by yosys or arachne-pnr ? 
I assume the latter in which case something as simple as different behavior of the random number generator could affect the SA result.

~/git/yosys ± git rev-parse --short HEAD
eb38722
~/git/arachne-pnr ± git rev-parse --short HEAD
a941129

~/git/swapforth ± git rev-parse --short HEAD
36f959d

jamesbowman, 2015-09-06:
Please can you pull and retry? I have reduced the size of the design so it is possible that it fits now.

polygonhell, 2015-09-06:
I'm on vacation this week, but I will try as soon as I an early next week.

Obijuan, 2015-09-07:
I had the same problem. I have just pull and try. Now it works nicely :-)  Thanks!

polygonhell, 2015-09-14:
I can confirm the issue is fixed.
Closing the issue.

closed 2015-09-14 by polygonhell