jamesbowman/swapforth #15: Can't connect using Python or Picocom when building from scratch

robot-army, 2015-10-14:
Hi there, 

I've tried building from scratch as per the reference document here: https://github.com/jamesbowman/swapforth/raw/master/j1a/doc/j1a-reference.pdf

The Verilog seems to build and program OK, but when trying to connect using the python terminal I get nothing. 

Note that using the j1a.bin as build in the repo works fine and gives me the correct response of:
Contacting... established
Loaded 208 words
>

I've tried checking out the last build that changed the j1a.bin file and building from scratch there, but it failed as well.

Colour me confused, any tips?

robot-army, 2015-10-14:
Tried reverting to revision 4f6ece5, that builds from scratch OK. Maybe it's something to do with the PLL implementation?

robot-army, 2015-10-14:
1e8708f works as well.

robot-army, 2015-10-14:
05cad81 and a956fe0 both won't build, failed to place...

jamesbowman, 2015-10-14:
Hmm, interesting. Just to confirm, this is with latest `yosys` and `arachne-pnr`? This is what I have.

```
$ cd yosys
$ git rev-parse --short HEAD
5dd3e93
$ cd ../arachne-pnr/
$ git rev-parse --short HEAD
efdb026
```

Also, I am building on 64-bit Linux.

When I go to 05cad81 and a956fe0 I see both fail to place with the above tools.

But the latest does place successfully, and does seem to be alive.

robot-army, 2015-10-15:
Thanks for getting back to me. 
Arachne-pnr was up to date, yosys wasn't, I was on ba4cce9. 
Did a pull then 
'''
make -j$(nproc)
sudo make install
'''
Cloned a fresh copy of swapforth:
'''
robotarmy@ubuntu:~/swapforth/j1a$ git rev-parse --short HEAD
7f24f57
'''
Then started to make and program:
'''
robotarmy@ubuntu:~/swapforth/j1a$ make -C icestorm
make: Entering directory '/home/robotarmy/swapforth/j1a/icestorm'
make -C ..
make[1]: Entering directory '/home/robotarmy/swapforth/j1a'
gforth cross.fs basewords.fs nuc.fs
tdp D50 tcp 0 python mkrom.py
make[1]: Leaving directory '/home/robotarmy/swapforth/j1a'
yosys  -q -p "synth_ice40 -top top -blif j1a.blif" j1a.v uart.v ../verilog/j1.v ../verilog/stack2.v
Warning: Wire top._j1.reboot has an unprocessed 'init' attribute.
Warning: Wire top._uart0._rx.hh has an unprocessed 'init' attribute.
Warning: Wire top.unlocked has an unprocessed 'init' attribute.
arachne-pnr -p j1a.pcf j1a.blif -o j1a.txt
seed: 1
device: 1k
read_chipdb +/share/arachne-pnr/chipdb-1k.bin...
  supported packages: tq144
read_blif j1a.blif...
prune...
read_pcf j1a.pcf...
instantiate_io...
pack...

After packing:
IOs          40 / 96
GBs          0 / 8
  GB_IOs     0 / 8
LCs          1187 / 1280
  DFF        712
  CARRY      74
  CARRY, DFF 7
  DFF PASS   102
  CARRY PASS 15
BRAMs        16 / 16
WARMBOOTs    1 / 1
PLLs         1 / 1

place_constraints...
promote_globals...
  promoted clk, 751 / 786
  promoted _j1.rstack.move, 305 / 305
  promoted _j1.dstack.move, 259 / 259
  promoted $techmap$techmap8870$auto$simplemap.cc:449:simplemap_adff$845.$logic_not$/usr/local/bin/../share/yosys/ice40/cells_map.v:12$8864_Y, 36 / 36
  promoted unlocked, 17 / 17
  promoted _j1.rstack.delta[1], 304 / 304
  promoted _j1.dstack.delta[1], 260 / 260
  promoted $techmap\_j1.dstack.$procmux$151_CMP, 16 / 16
  promoted 8 nets
    3 sr/we
    4 cen/wclke
    1 clk
  8 globals
    3 sr/we
    4 cen/wclke
    1 clk
realize_constants...
  realized 1
place...
  initial wire length = 10523
  final wire length = 4958
cell = 1391

After placement:
PIOs       29 / 96
PLBs       160 / 160
BRAMs      16 / 16

  place time 4.11s
route...
  pass 1, 19 shared.
  pass 2, 3 shared.
  pass 3, 0 shared.

After routing:
span_4     2585 / 6944
span_12    445 / 1440

  route time 2.25s
write_txt j1a.txt...
icebox_explain j1a.txt > j1a.ex
icepack j1a.txt j1a0.bin
icemulti -p0 j1a0.bin > j1a.bin
make: Leaving directory '/home/robotarmy/swapforth/j1a/icestorm'

robotarmy@ubuntu:~/swapforth/j1a$ make -C icestorm
make: Entering directory '/home/robotarmy/swapforth/j1a/icestorm'
make -C ..
make[1]: Entering directory '/home/robotarmy/swapforth/j1a'
gforth cross.fs basewords.fs nuc.fs
tdp D50 tcp 0 python mkrom.py
make[1]: Leaving directory '/home/robotarmy/swapforth/j1a'
yosys  -q -p "synth_ice40 -top top -blif j1a.blif" j1a.v uart.v ../verilog/j1.v ../verilog/stack2.v
Warning: Wire top._j1.reboot has an unprocessed 'init' attribute.
Warning: Wire top._uart0._rx.hh has an unprocessed 'init' attribute.
Warning: Wire top.unlocked has an unprocessed 'init' attribute.
arachne-pnr -p j1a.pcf j1a.blif -o j1a.txt
seed: 1
device: 1k
read_chipdb +/share/arachne-pnr/chipdb-1k.bin...
  supported packages: tq144
read_blif j1a.blif...
prune...
read_pcf j1a.pcf...
instantiate_io...
pack...

After packing:
IOs          40 / 96
GBs          0 / 8
  GB_IOs     0 / 8
LCs          1187 / 1280
  DFF        712
  CARRY      74
  CARRY, DFF 7
  DFF PASS   102
  CARRY PASS 15
BRAMs        16 / 16
WARMBOOTs    1 / 1
PLLs         1 / 1

place_constraints...
promote_globals...
  promoted clk, 751 / 786
  promoted _j1.rstack.move, 305 / 305
  promoted _j1.dstack.move, 259 / 259
  promoted $techmap$techmap8870$auto$simplemap.cc:449:simplemap_adff$845.$logic_not$/usr/local/bin/../share/yosys/ice40/cells_map.v:12$8864_Y, 36 / 36
  promoted unlocked, 17 / 17
  promoted _j1.rstack.delta[1], 304 / 304
  promoted _j1.dstack.delta[1], 260 / 260
  promoted $techmap\_j1.dstack.$procmux$151_CMP, 16 / 16
  promoted 8 nets
    3 sr/we
    4 cen/wclke
    1 clk
  8 globals
    3 sr/we
    4 cen/wclke
    1 clk
realize_constants...
  realized 1
place...
  initial wire length = 10523
  final wire length = 4958
cell = 1391

After placement:
PIOs       29 / 96
PLBs       160 / 160
BRAMs      16 / 16

  place time 4.11s
route...
  pass 1, 19 shared.
  pass 2, 3 shared.
  pass 3, 0 shared.

After routing:
span_4     2585 / 6944
span_12    445 / 1440

  route time 2.25s
write_txt j1a.txt...
icebox_explain j1a.txt > j1a.ex
icepack j1a.txt j1a0.bin
icemulti -p0 j1a0.bin > j1a.bin
make: Leaving directory '/home/robotarmy/swapforth/j1a/icestorm'

robotarmy@ubuntu:~/swapforth/j1a$ sudo iceprog icestorm/j1a.bin
init..
cdone: high
reset..
cdone: low
flash ID: 0x20 0xBA 0x16 0x10 0x00 0x00 0x23 0x12 0x67 0x21 0x22 0x00 0x95 0x00 0x28 0x04 0x11 0x11 0x99 0x46
file size: 32472
erase 64kB sector at 0x000000..
programming..
reading..
VERIFY OK
cdone: high
Bye.

'''
Then tried to connect with Python... it hung with no output for a while, pressed ^C and got the output (note that 'Contacting...' came out after pressing ctrl+C)
'''
robotarmy@ubuntu:~/swapforth/j1a$ sudo python shell.py -h /dev/ttyUSB1
^CContacting... Traceback (most recent call last):
  File "shell.py", line 66, in <module>
    swapforth.main(TetheredJ1a)
  File "../shell/swapforth.py", line 381, in main
    r.boot(image)
  File "shell.py", line 49, in boot
    self.reset()
  File "shell.py", line 42, in reset
    c = ser.read(1)
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 445, in read
    ready,_,_ = select.select([self.fd],[],[], self._timeout)
KeyboardInterrupt
'''

jamesbowman, 2015-10-15:
Oh, is this 32-bit or 64-bit?

robot-army, 2015-10-15:
```
robotarmy@ubuntu:~/swapforth/j1a$ uname -a
Linux ubuntu 3.19.0-30-generic #34-Ubuntu SMP Fri Oct 2 22:08:41 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```

64 Bit looks like to me. Admittedly, I'm running in VMware Fusion, but that shouldn't effect it? (also, I'm online right now, so if there's any checks you can recommend I'd be able to run them quickly!)

Thanks much!
Ryan

jamesbowman, 2015-10-15:
Yes, please do you get:
```
$ sha1sum build/nuc.hex 
b3370dea361fd65155a2bb5e2d764fb5065895cd  build/nuc.hex
```

robot-army, 2015-10-15:
Yep!
```
robotarmy@ubuntu:~/swapforth/j1a/build$ sha1sum nuc.hex
b3370dea361fd65155a2bb5e2d764fb5065895cd  nuc.hex
```

hmm....

robot-army, 2015-10-15:
Just to be sure, for a test, I've just cloned again, programmed the icestick and run the python shell. Here's what I get:

```
robotarmy@ubuntu:~/swapforth_2/j1a$ sudo iceprog icestorm/j1a.bin[sudo] password for robotarmy: 
init..
cdone: high
reset..
cdone: low
flash ID: 0x20 0xBA 0x16 0x10 0x00 0x00 0x23 0x12 0x67 0x21 0x22 0x00 0x95 0x00 0x28 0x04 0x11 0x11 0x99 0x46
file size: 32472
erase 64kB sector at 0x000000..
programming..
reading..
VERIFY OK
cdone: high
Bye.
(reverse-i-search)`python': sudo apt-get install ^Cthon3-serial
robotarmy@ubuntu:~/swapforth_2/j1a$ sudo python shell.py -h /dev/ttyUSB1
Contacting... established
Loaded 206 words
>
```

So, that works!

robot-army, 2015-10-15:
There's a difference in the bin files?

```
robotarmy@ubuntu:~/swapforth_2/j1a/icestorm$ sha1sum j1a.bin
a28aa8e405829742fc328cf4a3f100c21799f5c3  j1a.bin
robotarmy@ubuntu:~/swapforth_2/j1a/icestorm$ sha1sum ~/swapforth/j1a/icestorm/j1a.bin
3d630cde741fe74127b100da2bea0a416a024997  /home/robotarmy/swapforth/j1a/icestorm/j1a.bin
```

robot-army, 2015-10-15:
(swapforth_2 is the freshly cloned directory btw, swapforth is where I'm trying to build from scratch)

jamesbowman, 2015-10-15:
Right. At this point it looks like yosys/arache-pnr is where the difference lies. OK... puzzling.

robot-army, 2015-10-15:
Yep, also, my colleague B was having similar troubles. It's likely he was
using the same yosys version I pulled.

If you think of anything else you'd like me to check, or if you want to
shell in to have a look, I'm up for whatever!

robot-army, 2015-10-15:
@barnabyshearer was also suffering from this.

jamesbowman, 2015-10-15:
One thing... you are running a more recent Ubuntu than me.
What version Ubuntu do you have installed?

robot-army, 2015-10-15:
I'm on 15.04, what's yours? Happy to spin up another VM to try it?

jamesbowman, 2015-10-15:
Will do... I am on
Ubuntu 12.04.2 LTS

RGD2, 2015-11-26:
I'm on Ubuntu 14.04.3 LTS, with much the same problem... connecting after a clean build results in no action.

    dye025@dadc-nc:~/build/swapforth/j1a$ sha1sum icestorm/j1a.bin 
    4fad6215eda768073e397c8dd46772423586bb36  icestorm/j1a.bin

short HEAD's are:

- b69bda2 for swapforth
- 39337b7 for icestorm
- ab2d8e5 for yosys
- 788cf79 for arachne-pnr

When I drop the rebuild:

```
dye025@dadc-nc:~/build/swapforth/j1a$ git checkout -- icestorm/j1a.bin
dye025@dadc-nc:~/build/swapforth/j1a$ sha1sum icestorm/j1a.bin
a28aa8e405829742fc328cf4a3f100c21799f5c3  icestorm/j1a.bin
```

This one works...

RGD2, 2015-11-26:
Oh, just  a note: the first build works fine, it's the second one (after #flash build/nuc.hex, and rerunning make -C icestorm and reprogramming icestorm/j1a.bin ) that doesn't, for me at least. 
@robot-army Same for you?

RGD2, 2015-12-01:
Running with the master j1a.bin, adding some code, then using  ``#flash build/nuc.hex`` quitting and then ``make -C icestorm`` and ``sudo iceprog icestorm/j1a.bin`` results in a non-working image as well... just never connects. 

Shouldn't this procedure result in a new image with the added code included?

@jamesbowman : Should I make a new issue for this?

jamesbowman, 2015-12-01:
@RGD2 no new issue, thanks. These are all the same root cause. Under some circumstances the synthesis toolchain (arache-pnr, yoysys, icestorm) produces a bad FPGA image. The root cause is unclear.

RGD2, 2015-12-01:
Well, is there something I could do to help?

I'm newcomer to forth, but I already know I want swapforth in my labs, controlling my equipment.

robot-army, 2015-12-01:
Hey all, should be able to get back on this soon BTW. I haven't tested
under different Ubuntu builds yet.

R

RGD2, 2015-12-13:
This one's closed for me. @robot-army is it still broken at your end?

RGD2, 2015-12-27:
Works on Ubuntu 15.10

milkowski, 2016-08-20:
Hi, I'm experimenting with icestick & swapforth too, but on Windows with tools compiled using last mingw-w64-gcc-6.1.0 x64 toolchain. I'm using development versions of yosys and arachne-pnr form their respective master branches.
I noticed that the yosys synthesis is not deterministic and successive executions gives slightly different allocations of LCs which is also reported by arachne-pnr (I know that arachne-pnr annealing algorithm is intrinsically nondeterministic, but seeded to 1 by default and should be fixed - at least running the PnR alone does not differ between invocaions).
I'm curious if this is just my case or it is more common issue. As a side note, some of generated bitstreams that way have timing (or logic?) issues which prevent work of random on board leds. Simple re-generation of bitstream fixes the problem or changes the configuration of not working leds.

GuzTech, 2016-08-29:
I have the same issue. I'm running a fully updated Arch Linux and the latest checkouts of all the necessary tools and swapforth. No output when using the python shell here as well.

bmentink, 2016-09-28:
Hi, for no output running python shell I had to comment out:
```
        #waitcr()
        #ser.write(b'\r')
        #waitcr()
```
.. in reset() function. I am running latest Arch as well. I also found I had to run shell.py with python2 not python3, otherwise the #flash command would not work.

robot-army, 2016-09-28:
As soon as I get my icestick back I'll test!

GuzTech, 2017-02-26:
I've tried this with the latest Arch Linux, and with both Python 2 and 3 I get the expected output:
```
Contacting... established
Loaded 143 words
>
```
However, I do not see any output from commands. Here's an example:
```
>1 2 + .
  ok
>
```
Again, this is with both Python 2 and 3. The led blinking and easter examples also do not work. After trying a few commands, the system behaves as if I had written something else, like outputting the output of `words` when I execute something else.

> Hi, for no output running python shell I had to comment out:
> 
>         #waitcr()
>         #ser.write(b'\r')
>         #waitcr()
> .. in reset() function. I am running latest Arch as well. I also found I had to run shell.py with python2 not python3, otherwise the #flash command would not work.
> 
> See my reported issues here: [(https://github.com/jamesbowman/swapforth/issues/38)]

RGD2, 2017-04-21:
Is it locking up after the first newline in that blink example there? Does it work if you put the whole example on one line? (which should cause shell.py to send it as a line).

BrunoLevy, 2020-04-11:
To make it work, I had the same experience as robot-army:  
The shell was waiting without connecting,
By commenting the following three lines in shell.py, it connects and works as expected:
#waitcr()
#se.write(b'\r')
#waitcr()

Another issue I had was with the #flash command, still in shell.py, had to change one line in the serialize() function (last argument of ljust, binary string litteral instead of character that gave me an error, note: I'm using python 3)
 s = array.array('B', s).tostring().ljust(8192, b'\xff')