jamesbowman/swapforth #17: No documented way to run a word at boot

RGD2, 2015-12-01:
There doesn't seem to be a way to save an image with ``#flash `` whilst a word is running.
 
Is there any mechanism by which a system image can be made that will be running a custom application word from startup, yet still be able to be connected to and stopped/restarted via shell.py for further work if necessary? 

Without this, it's a bit difficult to deploy standalone, and to fix / debug after deployment.

jamesbowman, 2015-12-16:
So to make a flash image that turns the LEDs on at boot:

    :noname 31 leds quit ; init !
    #flash build/nuc.hex

To make a flash image that continuously blinks the LEDs

    : blink 0 begin dup leds 1+ 10 ms again ;
    ' blink init !
    #flash build/nuc.hex

To make a flash image that blinks the LEDs on power-up, but on reset will run the command prompt:

    :noname
        ['] quit init !
        blink
    ; init !
    #flash build/nuc.hex

RGD2, 2015-12-16:
Thanks @jamesbowman , Exactly what I was after.  :)
I didn't know about the init hook.

Can we have the latex source added under /doc or j1a/doc somewhere so nuggets like this can be added into the user manual? (ie, so I can help with that?)

The clear, beginner friendly documentation for swapforth and it's toolchain, combined with it's openness and sheer speed are killer features. ^_^

closed 2015-12-16 by RGD2

jamesbowman, 2015-12-19:
Yes, under j1a/doc/ is the manual source and the Makefile.