jamesbowman/swapforth #18: make -C j1a/icestorm Stops with Fatal Error : Failed to place

RGD2, 2015-12-07:
with Swapforth@9d45735 built on Ubuntu 14.04.3 LTS, with: cliffordwolf/yosys@5e1dc78 and cseed/arachne-pnr@eb7876b , 

    make -C j1a clean
    make -C j1a/icestorm

Results in a fatal error to place, apparently because 1297 of 1280 LCs are packed.

I think this an upstream issue with a recent change to yosys, but it's still our issue if we can no longer fit swapforth on a 1k :(

RGD2, 2015-12-08:
A problem with yosys?
I just found that it builds ok with 1199/1280 LCs on cliffordwolf/yosys@9f5b6e4 

... and *GOOD NEWS* the built image *doesn't* suffer #15 !
That is, this now works to permanently save developed code into the ROM:

    >: test 31 leds ;
    >#flash build/nuc.hex
    >#bye
    make
    make -C icestorm/
    sudo iceprog icestorm/j1a.bin
    sudo python shell.py -h /dev/ttyUSB1
    >test

Results in the leds all turning on.

Now, just need to figure out whether cliffordwolf/yosys@5e1dc78 has a regression, or not...
I'll open an issue.

cliffordwolf, 2015-12-08:
The following change to `j1a/icestorm/Makefile` will work around the issue (requires todays git head of yosys):

    -	yosys  -q -p "synth_ice40 -top top -blif j1a.blif" $(VERILOGS)
    +	yosys  -q -p "synth_ice40 -top top -abc2 -blif j1a.blif" $(VERILOGS)

See [yosys issue #109](https://github.com/cliffordwolf/yosys/issues/109) for details.

closed 2015-12-16 by RGD2