jamesbowman/swapforth #25: PIO0/PIO2 (HDR1/HDR2) access on icestick?

harwoodr, 2016-01-13:
I see in j1a.v that the pin headers on the board are defined... with addresses defined under "IO PORTS" but I can't seem to set them:

>255 $0020 io!
  ok
>$0020 io@ .
 255  ok
>255 $0010 io!
  ok
>$0010 io@ .
 27  ok

...am I missing something?

RGD2, 2016-01-15:
Confirmed.

Hmm...

RGD2, 2016-01-15:
It's a timing issue of some sort... there's a kludgy workaround:

    >-1 -1 $10 io! 
      ok
    >$10 io@ .
      255  ok

I hit the some issue with the j1a8k (with the leds on the hx8k breakout board) but it seemed to go away...

RGD2, 2016-02-09:
Seems fixed now with cliffordwolf/yosys@840a6dc, cseed/arachne-pnr@1a4fdf9, cliffordwolf/icestorm@b49d2d3, jamesbowman/swapforth@ef9aaa6.

```
sudo python shell.py -h /dev/ttyUSB1 -p ../common/
Contacting... established
Loaded 207 words
>-1 leds
  ok
>0 leds
  ok
>-1 $20 io!
  ok
>$20 io@ .
 255  ok
>-1 $10 io!
  ok
>$10 io@ .
 255  ok
```