jamesbowman/swapforth #26: j1a does not fit on the iCEstick

GuzTech, 2016-02-08:
Just as a heads up, using the very latest versions of icetools, arachne-pnr, and yosys, the process fails with:

```
yosys  -q -p "synth_ice40 -top top -blif j1a.blif" j1a.v uart.v ../verilog/j1.v ../verilog/stack2.v
Warning: Wire top._uart0._rx.hh has an unprocessed 'init' attribute.
arachne-pnr -p j1a.pcf j1a.blif -r -o j1a.txt
seed: 2531206741
device: 1k
read_chipdb +/share/arachne-pnr/chipdb-1k.bin...
  supported packages: cb121, cb132, cb81, cm121, cm36, cm49, cm81, qn84, swg16tr, tq144, vq100
read_blif j1a.blif...
prune...
read_pcf j1a.pcf...
instantiate_io...
pack...

After packing:
IOs          40 / 96
GBs          0 / 8
  GB_IOs     0 / 8
LCs          1309 / 1280
  DFF        711
  CARRY      73
  CARRY, DFF 8
  DFF PASS   103
  CARRY PASS 15
BRAMs        16 / 16
WARMBOOTs    1 / 1
PLLs         1 / 1

place_constraints...
promote_globals...
  promoted _bn00.RCLK, 751 / 786
  promoted _j1.rstack.move, 305 / 305
  promoted _j1.dstack.move, 259 / 259
  promoted $abc$9478$n767, 36 / 36
  promoted unlocked, 17 / 17
  promoted _j1.rstack.delta[1], 304 / 304
  promoted _j1.dstack.delta[1], 260 / 260
  promoted $abc$9478$n2252, 16 / 16
  promoted 8 nets
    3 sr/we
    4 cen/wclke
    1 clk
  8 globals
    3 sr/we
    4 cen/wclke
    1 clk
realize_constants...
  realized 1
place...
fatal error: failed to place: placed 1121 LCs of 1310 / 1280
Makefile:7: recipe for target 'j1a.bin' failed
```

combinatorylogic, 2016-02-08:
Adding '-r' option to arachne-pnr usually helps.

GuzTech, 2016-02-08:
If you look at the third line of the output, you can see that I did that in hopes that it would help but unfortunately it didn't. The original Makefile does not have it, so it will definitely not work as it is now. I suspect it's caused by icetools, since I did build j1a in the past correctly.

combinatorylogic, 2016-02-08:
Did not notice. Although it worked for me - you may have to try more than once.

GuzTech, 2016-02-09:
I've tried over a hundred times now, but I always get the same output (except for the randomized seed). But isn't it normal that it cannot P&R it? There are 1310 LCs it wants to P&R, but there are only 1280 LCs, so no matter how you slice it, it's not going to fit. Either that, or I'm just very unlucky.

RGD2, 2016-02-09:
I get 1194 LCs with `(cd swapforth/j1a; make clean; make; make bootstrap; make; make j1a; make connect)`
This with tools versions [using `(git rev-parse --short HEAD)`]: {yosys, arachne-pnr, icestorm, swapforth} = {840a6dc, 1a4fdf9, b49d2d3, ef9aaa6}

The number of LC's you were getting sounds about right for an initial build sans swapforth.fs or the extra abc2 flag  - it changes once the swapforth image is "baked in" after executing `#flash build/nuc.hex` on the j1a and then running `make` again to regenerate `build/ram.v`

I think you will also find that you need to run yosys with the `-abc2 ` optimisation flag. (see #18 )

But `make j1a` or even just `make -C icestorm j1a.bin` from within `swapforth/j1a` ought to work.

Any luck?

combinatorylogic, 2016-02-09:
It's really weird. I reproduced the same behaviour on Ubuntu 14.04, but it does not fail to place and route on OS X, with all the same checkouts.

GuzTech, 2016-02-09:
Ok, so when I do what RGD2 does and call make from the j1a folder, then I get the same output and everything works just fine. However, what I did was call make from within the j1a/icestorm folder, and then it gives the same output as above and fails.

closed 2016-02-10 by GuzTech