jamesbowman/swapforth #27: make j1a fails to place

RGD2, 2016-02-25:
Just noticed this now. This is critical (it blocks making application images) but there is a work around to get a compiled, bootstrapped image.

With swapforth@efb3813 ,  cliffordwolf/icestorm@f2b2549 , cseed/arachne-pnr@1a4fdf9 and cliffordwolf/yosys@22c549a 

```

~/build/swapforth/j1a$ make j1a
make -C icestorm j1a
make[1]: Entering directory `/home/likewise-open/NEXUS/dye025/build/swapforth/j1a/icestorm'
yosys  -q -p "synth_ice40 -top top -abc2 -blif j1a.blif" j1a.v uart.v ../verilog/j1.v ../verilog/stack2.v
Warning: Wire top._uart0._rx.hh has an unprocessed 'init' attribute.
arachne-pnr -p j1a.pcf j1a.blif -o j1a.txt
seed: 1
device: 1k
read_chipdb +/share/arachne-pnr/chipdb-1k.bin...
  supported packages: cb121, cb132, cb81, cm121, cm36, cm49, cm81, qn84, swg16tr, tq144, vq100
read_blif j1a.blif...
prune...
read_pcf j1a.pcf...
instantiate_io...
pack...

After packing:
IOs          40 / 96
GBs          0 / 8
  GB_IOs     0 / 8
LCs          1219 / 1280
  DFF        711
  CARRY      73
  CARRY, DFF 8
  DFF PASS   103
  CARRY PASS 15
BRAMs        16 / 16
WARMBOOTs    1 / 1
PLLs         1 / 1

place_constraints...
promote_globals...
  promoted _bn00.RCLK, 751 / 786
  promoted _j1.rstack.move, 305 / 305
  promoted _j1.dstack.move, 259 / 259
  promoted $abc$10779$n1230, 36 / 36
  promoted unlocked, 17 / 17
  promoted _j1.rstack.delta[1], 304 / 304
  promoted _j1.dstack.delta[1], 260 / 260
  promoted $abc$10779$n1158, 16 / 16
  promoted 8 nets
    3 sr/we
    4 cen/wclke
    1 clk
  8 globals
    3 sr/we
    4 cen/wclke
    1 clk
realize_constants...
  realized 1
place...
fatal error: failed to place: placed 1121 LCs of 1220 / 1280
make[1]: *** [j1a.bin] Error 1
make[1]: Leaving directory `/home/likewise-open/NEXUS/dye025/build/swapforth/j1a/icestorm'
make: *** [j1a] Error 2
```

The work around is to run `make -C swapforth/j1a bootstrap` first, after which `make -C swapforth/j1a j1a` works, but since this make a 'fresh bootstrapped ' 207-word swapforth nuc.hex, there isn't any way to use `#flash build/nuc.hex` to make an application image, since the same failure-to-place occurs there too.

:(

RGD2, 2016-02-27:
With swapforth@5951611 and cliffordwolf/yosys@c89f61c , under Mac OS X:
It sometimes works both ways, and sometimes doesn't... just have to keep rerunning `make -C j1a/icestorm j1a.bin` until it works.

jamesbowman, 2016-03-08:
OK, after a discussion in cliffordwolf/yosys#23, the latest yosys has a couple of tweaks that help placement a lot.  There is also a recent patche to the J1a itself that shrinks its size by about 20 LCs or so.

It should place reliably now.
```
$ arachne-pnr -v
arachne-pnr 0.1+136+1 (git sha1 1a4fdf9, g++ 6.0.0 -O2)
$ yosys -V
Yosys 0.6+6 (git sha1 e7ed653, gcc 6.0.0 -fPIC -Os)
```

RGD2, 2016-03-08:
Fixed! That did the trick!

closed 2016-03-08 by RGD2