jamesbowman/swapforth #30: How do you do multi-line input with the shell?

kasbah, 2016-05-10:
Hey, first time using any sort of Forth so this might be a stupid question. I arrived here by trying to control the LEDs on my new ICE40 dev kit with the least effort possible. I managed to do that. Thanks so much for paving the way! 

I am a bit stuck now though as I haven't been able to figure out how to program in the multi-line words mentioned in the [reference manual](https://raw.githubusercontent.com/jamesbowman/swapforth/master/j1a/doc/j1a-reference.pdf) and [your video](https://www.youtube.com/watch?v=rdLgLCIDSk0), I tried `Shift-Enter` an `Alt-Enter` and `Ctrl-V` and `;` but none of those seem to work.

jamesbowman, 2016-05-10:
`Enter` is the continuation character -- here I'm entering the 'blink' program.

[image]

Swapforth should do something to show that you're in compilation state, so the interaction looks like:
```
> : blink
    32 0 do
      i leds
      100 ms
    loop
    ;
ok
> blink
```

larsbrinkhoff, 2016-05-10:
I suggest something like

    : ok   state @ 0= if ."  ok" cr then ;

kasbah, 2016-05-10:
Ah, got it running. The interaction looks like the image you linked. What do you mean by it "_should_ do something to show you're in compilation state"? Is there something wrong with the way I have run things?  If not, would it be worth mentioning somewhere in the docs, for newbies like me, that you just hit enter and that the "ok" response is expected? The interaction in the video looks different with a "+" character at the start of new lines. 

@larsbrinkhoff I copy pasted your code into the interpreter and it doesn't seem to have any effect.

jamesbowman, 2016-05-10:
@kasbah I mean "I should fix this so it is less confusing" ... the code that @larsbrinkhoff posted is intended to go into the Forth interpreter.