jamesbowman/swapforth #37: j1a IO is broken

RGD2, 2016-08-22:
This is on swapforth@9157271 , cseed/arachne-pnr@52e69ed , and cliffordwolf/yosys@7a33b98 

Symptoms: LED's on iceStick (and possibly other IO ports) don't set/clear properly. 

Steps to reproduce (with iceStick plugged in):

```
cd swapforth/j1a
make clean
make 
sudo make bootstrap 
sudo make j1a connect
-1 leds
```

After this, one led is not lit, but `-1 -1 leds drop` will set all, likewise `-1 0 leds drop` will leave that one led on.

The affected .bin, .blif and .txt are here:
[broken_j1a.zip](https://github.com/jamesbowman/swapforth/files/429530/broken_j1a.zip)

RGD2, 2016-08-22:
Also under cliffordwolf/yosys@21e1bac on arm.

RGD2, 2016-08-22:
the obvious kludgy workaround is `: leds dup leds drop ;` , but this problem hints at deeper trouble.

bmentink, 2016-09-29:
..this is still broken, also reads from led port, turns random led's on ...

RGD2, 2016-09-29:
Please save and upload the j1a.blif and .bin.

This seems to happen only sometimes, and I'd like to get to the bottom of
it.

bmentink, 2016-09-29:
Here is a link to the files: [https://www.dropbox.com/s/fjesc5dve9iakig/j1a.zip?dl=0](url)
Usually doing -1 leds, turns on only 4 led's .. -1 -1 leds drop fixes ..

bmentink, 2016-10-02:
I have just updated yosys to  r2983.4eb0d6f-1 and arachne-pnr to  r171.52e69ed-1 and the LED's seem to work just fine now  ( This is latest builds from AUR on ArchLinux)

RGD2, 2016-11-25:
Yosys: `git rev-parse --short head` gives `5c2c78e`.

aand it's fixed here too. Closed.

closed 2016-11-25 by RGD2