jamesbowman/swapforth #38: shell.py issues on ArchLinux

bmentink, 2016-09-27:
1st issue: I had to comment out the following section in shell.py to even get it running:

> def reset(self, fullreset = True):
        ser = self.ser
        ser.setDTR(1)
        if fullreset:
            ser.setRTS(1)
            ser.setRTS(0)
        ser.setDTR(0)
        
        def waitcr():
            while ser.read(1) != chr(10):
                pass

        #waitcr()
        #ser.write(b'\r')
        #waitcr()

2nd issue: bye does not exit shell ... just get:

> >bye
 ?+

3rd issue:  #flash command does not work:

> >#flash build/nuc.hex
please wait...
Traceback (most recent call last):
  File "./shell.py", line 72, in <module>
    swapforth.main(TetheredJ1a)
  File "../shell/swapforth.py", line 385, in main
    r.shell()
  File "../shell/swapforth.py", line 339, in shell
    self.shellcmd(cmd)
  File "../shell/swapforth.py", line 197, in shellcmd
    d = self.serialize()
  File "./shell.py", line 68, in serialize
    s = array.array('B', s).tostring().ljust(8192, chr(0xff))
TypeError: ljust() argument 2 must be a byte string of length 1, not str

Seems shell.py is a bit broken .. also the documentation says to connect on /dev/ttyUSB0, but on my system it is /dev/ttyUSB1, USB0 is the programmer SPI interface ..

RGD2, 2016-09-28:
Just `make` By itself without `-C icestorm` should be enough to rebuild build/ram.v 

But yes, it is needed after a #flash

bmentink, 2016-09-28:
Thanks, any comment on issue #1

closed 2017-11-08 by jamesbowman