jamesbowman/swapforth #39: Ramping up the clock rate

bmentink, 2016-09-29:
Not an issue as such, but due to the lack of a forum will post here.

I would like to run the clock much higher to see what the limit is to run swapforth at.
I tried to het a handle on this by implementing a  bare minimal PWM verilog module and ramped up the PLL until the PWM stopped working.
This went ok till about 550Mhz, (2.2MHz PWM at 8 bits) which I thought was pretty awesome ..

As a 1st attempt with swapforth, I doubled the PLL to 96Mhz  (7 as the .DIVF instead of 3) and in the uart.v module doubled the CLKFREQ constant to keep the baudrate at 460800 baud. However it does not work. I am not sure if the problem is the uart or just running that fast ... any idea's

RGD2, 2016-09-29:
If you disable the random number generator in icestorm/j1a.v, then icetime
makes better sense of it. The loop in that part causes icetime to give
silly results.

bmentink, 2016-09-29:
Are you meaning the section marked as "Ring Oscillator"? That's the only reference to random I can see .. and what is icetime?

By the way, where are all the modules starting SB_* defined? I would especially like some documentation on those .. I am guessing they are the technology modules for the FPGA ?..

RGD2, 2016-09-30:
Yes, that's it.

Look up icetime [on Clifford's site](http://www.clifford.at/icestorm/). 
Scroll down to the "Where is the Documentation?" heading, there's a link there to the [Lattice iCE Technology library](http://www.latticesemi.com/~/media/LatticeSemi/Documents/TechnicalBriefs/SBTICETechnologyLibrary201504.pdf), which documents all those SB_* things. I'm not actually sure where they're implemented in verilog - it would have to be somewhere in the yosys/icestorm/arachne-pnr toolchain.

bmentink, 2016-09-30:
Tried disabling the Ring Oscillator, but still can't run the PLL at 96Mhz ..
.. no response from shell.py .. or minicom

jamesbowman, 2016-09-30:
Maybe try leaving the uart unchanged and connecting at 921600?

bmentink, 2016-10-01:
Tried that, same result. Maybe there is something wrong with the timing at that speed, I don't know enough to do a timing anaylsis ..

bmentink, 2016-10-18:
Been a couple of weeks now ... any comment @jamesbowman 
By the way James, excellent Forth, works very well  ......