jamesbowman/swapforth #43: simulator (verilator) does not compile

bmentink, 2016-10-12:
I tried running "make bootstrap" for j1a .. but it fails ..

```
make bootstrap
make -C verilator
make[1]: Entering directory '/home/bmentink/Builds/SwapForth/swapforth/j1a/verilator'
Makefile:33: warning: overriding recipe for target '../build/nuc.hex'
Makefile:12: warning: ignoring old recipe for target '../build/nuc.hex'
verilator -Wall --cc -DVERILATOR=1 j1a.v ../verilog/*.v --top-module j1a --exe sim_main.cpp
make -C obj_dir CXXFLAGS="-fPIC" OPT_FAST="-O2" -f Vj1a.mk Vj1a
make[2]: Entering directory '/home/bmentink/Builds/SwapForth/swapforth/j1a/verilator/obj_dir'
g++ -fPIC -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_TRACE=0 -DVM_COVERAGE=0 -Wno-char-subscripts -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable      -O2 -c -o sim_main.o ../sim_main.cpp
../sim_main.cpp: In function ‘int main(int, char**)’:
../sim_main.cpp:23:12: error: ‘class Vj1a’ has no member named ‘v__DOT__ram_prog’; did you mean ‘j1a__DOT__ram_prog’?
       top->v__DOT__ram_prog[i] = v;
            ^~~~~~~~~~~~~~~~
At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-parentheses-equality’
make[2]: *** [Vj1a.mk:65: sim_main.o] Error 1
make[2]: Leaving directory '/home/bmentink/Builds/SwapForth/swapforth/j1a/verilator/obj_dir'
make[1]: *** [Makefile:9: obj_dir/Vj1a] Error 2
make[1]: Leaving directory '/home/bmentink/Builds/SwapForth/swapforth/j1a/verilator'
make: *** [Makefile:8: bootstrap] Error 2
```

combinatorylogic, 2016-10-13:
Verilator version?

jamesbowman, 2016-10-13:
Right.. seems most likely.

My Verilator is 3.856 and after the build I get:

    $ grep __DOT__ram_prog verilator/obj_dir/Vj1a.h 
            VL_SIG16(v__DOT__ram_prog[4096],15,0);

bmentink, 2016-10-13:
The version giving the error is:

Verilator 3.886 2016-07-30 rev verilator_3_884-13-gb4a7e46

The grep above gives the same result for me ..

bmentink, 2016-10-16:
Can I give an other help?

RGD2, 2016-10-17:
Mine's Verilator 3.856 2014-03-11 rev verilator_3_855-19-g749ff02
Which works. If you're on debian you can downgrade to it - I'm just using
the prebuilt version ubuntu/debian has.
*Possibly* worth raising as a regression bug against Verilator - since the
newer version breaks our code.

combinatorylogic, 2016-10-19:
There is a way to get the old naming back: just add an option `--l2-name v` to verilator. Did not try with the older versions though.

jamesbowman, 2016-10-19:
Just tried it with my old Verilator (3.856):

    %Error: Invalid Option: --l2-name

jamesbowman, 2016-10-19:
Looks like this changed in 3.884:

http://www.veripool.org/projects/verilator/news

closed 2017-10-24 by jamesbowman