jamesbowman/swapforth #44: Is this project dead .. or just sleeping

bmentink, 2016-10-24:
Hi Guys,

Not to create any offence. But is this project actively worked on? I see no push's for 5 months, and no current issues worked on.

I am happy to help out, at least with finding bugs . 
I hope this project is not dead, as I think it has a lot of promise. 

Currently I am using it in my "Electronic Drum Brain" project, where the main requirements are writing Verilog peripherals and interfacing them to SwapForth side. (A proper io address decoder here would be useful). The Forth code will tie it all together and send Midi commands to the host PC which has a drum synthesiser  ...

jamesbowman, 2016-10-24:
Sleeping!

jamesbowman, 2016-10-24:
If you have a specific bug against the code base, please open the issue.

closed 2016-10-24 by jamesbowman