jamesbowman/swapforth #47: Feature Request: Support for the ICOBOARD

bmentink, 2017-03-14:
Any chance of support for this board? Development is usually by attachment to a RPi, then SSH into it from your PC. Would it be possible to have a virtual console to interact with Forth this way?
Programming sram/flash should work out-of-the-box.

(The connection between the two is SPI through a Linux kernel driver, but not sure if both ways is implemented ..)

RGD2, 2017-04-20:
Something I want also. 

The support could be done much the same way the j1a8k and j4a, with just a change of .pcf file, but you'd need to add some way to get rs232 + the reset control line via the hat interface, or else change the j1a.v to a version that talks over SPI instead, and change shell.py to handle talking over that interface -- OR, add a linux kernel driver for a /dev/ttyS which actually lives on the fpga. 

There is some work in the latter direction already, specifically for the ras Pi:  see https://opentechlab.org.uk/videos:002:notes 

This seems to be intended at being able to add arbitrary interfaces (not just serial, but anything else as well) to the fpga, and allowing linux to consider those devices the same as any other plug-in card.

This would be super powerful for the ras Pi if the high-bandwidth camera/display interfaces could be abused as high bandwidth links to/from an fpga. (perhaps, appears theoretically possible, maybe).

You could then plug proper gBit ethernet + sata into a raspi and get  far more bandwidth than is currently possible over the raspi's only link to the outside world otherwise: Its one USB2.0 hi-speed root port (which is only about half as fast as it ought to be - really struggles to hit 180 Mbit/s at best).

Currently, I'm using a linux SBC with real gigE, and just plugging a hx8k breakout board in via usb.

jamesbowman, 2017-10-24:
If anyone sends a pull request with ICOBOARD support, I'll merge it.
Closing.

closed 2017-10-24 by jamesbowman