jamesbowman/swapforth #55: I have ported j1b to Artix-7 (cmod-A7 board and Arty)

bmentink, 2017-10-24:
Hi James,

I don't know if you are still contactable or if this project is still active, but I have posted this just in case are still interested.

I have j1b running at 80Mhz on both Spartan-6 and Artix-7 boards - it will run at 100Mhz (just) on the Artix-7, but borderline as there is timing issues on the ram dstack  .... Is there a way to improve this?

Also for your interest, I have implemented a number of Floating point hardware modules in verilog and interfaced them to Forth (F+,F-,F*,F/ and F-inverse-sqrt) They all happen in less than 3 clock cycles.

On Artix-7 Forth + FP takes up 3500 LUT's .......

jamesbowman, 2017-10-24:
Sounds interesting - will you be publishing the source?

I don't know of any ways to improve j1b's timing. As I recall I'd already tried to optimize its paths as much as possible.

bmentink, 2017-10-24:
I am happy to provide the source, when I get it all together and documented.

However, this should be a two-way street, you are very hard to contact. If I am to donate the code, then  I would expect that to change .... ;)  (I have asked you a number of questions in the past and have been ignored ..)

jamesbowman, 2017-10-24:
You are welcome to fork the repo, then you can decide how much time to devote to supporting an open-source project.

OK to close this issue?

bmentink, 2017-10-24:
Ok, get the hint. You don't want to put in any more time :)
I will contribute to an existing fork of your code that is well supported ..

Cheers, and have a great day.

PS: Close away ....

closed 2017-10-24 by jamesbowman