jamesbowman/swapforth #56: ram.v for ice40UP5

igor-m, 2017-11-08:
Hi, I've got an ice40UP5 board and want to try with J1a(b). 
Is there any chance to get available a ready "ram.v" with the basic words set? 
My current understanding is the ram.v is the same source for all 1k/4k/8k/UP5 chips. 
My intention is to get it working with IceCube2 under Win.
Thanks

jamesbowman, 2017-11-08:
Hmm, am having trouble looking up the ice40UP5 - do you have a link?

igor-m, 2017-11-08:
http://www.latticesemi.com/en/Products/FPGAandCPLD/iCE40Ultra.aspx
ice40UP5k - at the right side of the table

jamesbowman, 2017-11-09:
Correct, the RAM contents should be possible.

With the correct J1a implementation in Verilog (including the UART pins hooked up) this should have a full set of words. It is a 4Kx16 RAM, contents in hex:
[nuc.hex.txt](https://github.com/jamesbowman/swapforth/files/1456112/nuc.hex.txt)

Then you will want to replace this file:

https://github.com/jamesbowman/swapforth/blob/master/j1a/icestorm/j1a.v

with a version for IceCube2. Keep the j1, the UART, and hook up the RAM initialized with the above file.
The IO ports won't apply to your board, so delete all that.

Hope this helps, J.

igor-m, 2017-11-09:
Would not be easier to have the "ram.v" ? You need it in order to compile the j1a in Icecube2, afaik:
j1.v + j1a.v + ram.v

closed 2017-11-22 by igor-m