jamesbowman/swapforth #59: nandland.com   go board

oystercatcher, 2018-06-03:
I cloned  swapforth  and have been looking at j1a.   There is a subdirectory
called  nandland-go  which has a bin file that I loaded into the go board.
The 7seg displays  show 88.    I tried using  python3  shell.py  -h  /dev/ttyUSB1          
but the command always hangs  output similar to previous reports.

I am able to connect to the go board  with gtkterm  at baud 115200   
  0 leds  turns off the 7 seg displays    and  words  gives a list

I ran the Makefile  and compared its output with the supplied  j1a.bin 

the results are different as my file is smaller and after connecting with
gtkterm   I see that words  is much smaller  maybe 145 instead of 
around 200.    

My system is fedora 27  
yosys -V
Yosys 0.7 (git sha1 UNKNOWN, gcc 7.1.1 -fPIC -Os)

I  presume I am missing something in my setup  for  yosys 
as I am  new owner of the go board and have been able to
get some small programs running.

oystercatcher, 2018-06-04:
I  found  another project on sourceforge that I think is more current
https://sourceforge.net/projects/mecrisp/

I also noticed an error in  yosys   when  I ran  read_verilog on  ../build/ram.v

RGD2, 2018-06-04:
Things to try: it’s recommended to see if the latest git builds of the
icestorm toolchain fix your problems, since they are evolving so rapidly.
Follow the installation steps on Clifford’s project icestorm website. Main
thing is to have the dependencies, but I’ve been able to get swapforth /
j1a working on even macosx with homebrew and macports. It also works on
SBC’s with Ubuntu or Debian (odroid-c2, xu4, maybe raspi 3), so long as you
have at least a gig of ram.

Also, low word count can also be due to an error during the post-bootstrap
part of the swap forth build- ie, where #include swapforth.fs happens.
If there’s a serial transmission error, the only hint is the one line with
a question mark the at beginning - the rest of the upload (#include is
executed within the python shell.py script, it just uploads the mentioned
file) doesn’t notice and plows on.

I usually do that step using verilator, and then do a final rebuild to
include the new ram.v contents. (The newer versions of icestorm now support
changing just the ram contents of a bit file, without having to do a full
yosys /Arachne-pnr rebuild, but we haven’t integrated that into swapforth
yet, AFAIK)

Mecrisp-ice is very good if you want a more complete forth system, but will
require a 4K/8K chip to fit its larger memory. (I think the nandland go is
a 1K ice40?).

What is it you would like to do with swapforth? Do you have a specific
application in mind?

oystercatcher, 2018-06-05:
I have tested the binary from   swapforth    as well as the one from  Mecrisp - Native code Forth for MSP430 and ARM Cortexfor the nandland.com  go  board.    
The binary from  mecrisp   uses   picocom    called terminal  in their setup and it seems to work.
Having a  cli  to the  p1a   is  amazing.              
I don't know forth   but I remember that it was the os for the  solaris   boot rom  Thank you  for making forth  new again.

oystercatcher, 2018-06-05:
Thanks for the information as I came to this project as a novice at  pretty much everything
around fpga's  and  forth.   I am interested in learning about the processor design and how 
it was implemented in verilog and icestorm.   

Is it possible that my build of  j1a.bin is ok but the failed connection with python3 shell.py  /dev/ttyUSB1  prevents an update that provides  the additional words.

I am going back to try the build from swapforth using picocom  and see if I can do
a transfer for that include file.

Sorry I don't quite understand the process as yet but I think I get the idea that
after the transfer of the include file completes;  forth will have the added words
and then it is possible to create  a usable  j1a.bin file.

jamesbowman, 2018-06-05:
If I recall, the nandland does not reset on UART connection.
So maybe try connecting with ``python3 shell.py /dev/ttyUSB1`` then while it's hanging hit reset on the board?

oystercatcher, 2018-06-12:
I  have tested the binary from     https://sourceforge.net/projects/mecrisp/files/mecrisp-ice-1.3.tar.gz
and it is working with a cli and forth.    I also tested their toolchain  and had a working binary on the
nandland go board.    

Lots to study.

closed 2018-06-13 by oystercatcher