jamesbowman/swapforth #61: What's EASE_IO_TIMING for?

redfast00, 2019-02-02:
I'm learning to program for FPGA's, and read through the source of the j1 to learn. I wonder what EASE_IO_TIMING is for [here](https://github.com/jamesbowman/swapforth/blob/master/j1a/icestorm/j1a.v#L209). What function does it have/what problem does it solve?

RGD2, 2019-02-03:
Excellent question. It’s an example of pipelining. (Also of how defines are
used in verilog).

Pipelining gives the io subsystem more time per clock cycle by registering
the inputs to it from the j1. These inputs otherwise are “late” by the time
they take to be decoded from the instruction, meaning less time for the io
subsystem logic delay.

Probably yields a higher safe clock rate. Which can be confirmed now that
the icestorm tools have static timing analysis (they didn’t when that was
written, So easily testing with/without by commenting the define line was
probably helpful).

Putting the registers in there delays response from the io subsystem by a
clock cycle (bad, potentially, but dealt with elsewhere) but means that
these signals are “fresh” with the start of each clock cycle, giving the io
subsystem as much time as possible (good).

The `define / `else / `endif part is Probably no longer necessary
ultimately (would remove the define and the uncompiled code from else to
endif), but where you’d like to compile optionally different parts in code
with minimal changes or perhaps just different Makefile targets, this is
how you’d do it.

If the “define” line isn’t there or is commented, one can “turn it on” from
the make file by adding it to the call to yosys, so with/without testing is
easier.

Clearly James found it more helpful than not, so that define was simply
left enabled.

iIRC io reads have an extra cycle added in the j1 core to account for the
extra cycle delay before the data in from the io subsystem is ready, so it
will probably add a bug if disabled now anyway.

Memory “reads” also end up taking two clock cycles per instruction, but
that can’t be avoided on the ice40 architecture, because unlike the
spartan6, the ram blocks are only pseudo dual port - so one read, one
optional write, rather than one read and one read or write. But it gets a
bit murkier there because of the odd timing the ice40’s ram blocks have for
their control signals (refer to using ice40 ram documents - I can’t recall
the details other than it wanting odd timing cycle to cycle for writes).

Hope that helps :)

jamesbowman, 2019-02-03:
Excellent explanation, thanks @RGD2 !

redfast00, 2019-02-04:
Thank you @RGD2

closed 2019-02-04 by redfast00