jamesbowman/swapforth #86: Converting j1a to use ICEbreaker

arblake, 2024-01-12:
Hi, I am trying to adapt J1a to the Icebreaker board which uses the ICE40 UP5K chip. This is the error:
ERROR: Cell 'pclk$sb_io' cannot be bound to bel 'X7/Y0/io0' since it is already bound to cell '_hdr1.io[5]._io'

RGD2, 2024-01-13:
Copy the  j1a.pcf file to a new named one specific to the board you’re
using, and do similar renames throughout the Makefiles to generate a new
build using that new pin constraint file: every different board needs a
different one.

The error is from trying to assign two signals to the same physical pin.
And you shouldn’t fix it by changing the original file, which is for the
icestick.
You can look at what I did for the j1a-8k: basically you could copy all the
Makefile entries for that, just changing “8k” for “up5k”, but you’ll need
to delete all the specific PIN numbers after copying the PCF, then read
through your board’s schematic to put the right assignments in for your
board.

-- Remy

arblake, 2024-01-14:
Thanks for your help. I am learning , mainly by doing. Now I have these error messages.
Warning: IO 'PIO2_12' is unconstrained in PCF and will be automatically placed
Warning: IO 'PIO2_11' is unconstrained in PCF and will be automatically placed
Warning: IO 'PIO1_06' is unconstrained in PCF and will be automatically placed
Warning: IO 'PIO1_05' is unconstrained in PCF and will be automatically placed
Warning: IO 'PIO0_06' is unconstrained in PCF and will be automatically placed
Warning: IO 'PIO0_05' is unconstrained in PCF and will be automatically placed
Warning: IO 'resetq' is unconstrained in PCF and will be automatically placed
Warning: IO 'PIO1_20' is unconstrained in PCF and will be automatically placed
Warning: IO 'PIO1_19' is unconstrained in PCF and will be automatically placed
Warning: IO 'PIO1_18' is unconstrained in PCF and will be automatically placed
Warning: IO 'D5' is unconstrained in PCF and will be automatically placed
Warning: IO 'D4' is unconstrained in PCF and will be automatically placed
ERROR: PLL 'uut' couldn't be placed anywhere, no suitable BEL found. Did you mean to use a PAD PLL ?
    PLL bel 'X12/Y31/pll_3' cannot be used as it conflicts with input 'pclk$sb_io' on pin '35'.

ERROR: cell type 'SB_PLL40_CORE' is unsupported (instantiated as 'uut')
37 warnings, 2 errors
See attached files for additional information.

[j1aup5k.nplog.txt](https://github.com/jamesbowman/swapforth/files/13930901/j1aup5k.nplog.txt)
[j1aup5k.pcf.txt](https://github.com/jamesbowman/swapforth/files/13930902/j1aup5k.pcf.txt)
[j1aup5k.v.txt](https://github.com/jamesbowman/swapforth/files/13930903/j1aup5k.v.txt)

make: *** [Makefile:23: j1aup5k.asc] Error 255

arblake, 2024-01-14:
Thanks for your help. I am learning by doing ...... and failing, but slow progress. The two devices are quite different.

arblake, 2024-01-15:
It all builds and I can upload to my device. I connect to a serial port and there is "communication" but it looks like the baud rate is wrong. uart.v uses a clock, using the variable clk. clk is generated by the PLL in my main file j1aup5k.v and it is 40MHz but the uart.v has comments about the 12MHz hardware clock. thus I am confused.

arblake, 2024-01-17:
Success, I can get an " ok " prompt and " words" lists the words. BUT 1 2 +  .   only gives ok prompt no answer. 
I can only communicate using GTK term. The suggested shell.py script does not talk to the serial port and it is not a permissions problem.

closed 2024-01-18 by arblake